node-hid/node-hid #550: Same system path for different usagePages on macOS

JeremyXXXuuu, 2024-10-24:
## Issue Description:
I encountered an issue when working with HID devices on macOS where different usagePages are assigned the same system path. This seems to be a problem, as node-hid cannot distinguish between them, potentially leading to incorrect device detection or usage.

## Example:
Here is an example of the devices returned by node-hid:
```
[
  {
    "vendorId": 13606,
    "productId": 770,
    "path": "DevSrvsID:4295921431",
    "serialNumber": "TK24C00003",
    "manufacturer": "--",
    "product": "Dongle Evo USB",
    "release": 513,
    "interface": 0,
    "usagePage": 12,
    "usage": 1
  },
  {
    "vendorId": 13606,
    "productId": 770,
    "path": "DevSrvsID:4295921431",
    "serialNumber": "TK24C00003",
    "manufacturer": "--",
    "product": "Dongle Evo USB",
    "release": 513,
    "interface": 0,
    "usagePage": 11,
    "usage": 5
  },
  {
    "vendorId": 13606,
    "productId": 770,
    "path": "DevSrvsID:4295921433",
    "serialNumber": "TK24C00003",
    "manufacturer": "--",
    "product": "Dongle Evo USB",
    "release": 513,
    "interface": 1,
    "usagePage": 65280,
    "usage": 1
  },
  {
    "vendorId": 13606,
    "productId": 770,
    "path": "DevSrvsID:4295921433",
    "serialNumber": "TK24C00003",
    "manufacturer": "--",
    "product": "Dongle Evo USB",
    "release": 513,
    "interface": 1,
    "usagePage": 65280,
    "usage": 3
  },
  {
    "vendorId": 13606,
    "productId": 770,
    "path": "DevSrvsID:4295921433",
    "serialNumber": "TK24C00003",
    "manufacturer": "--",
    "product": "Dongle Evo USB",
    "release": 513,
    "interface": 1,
    "usagePage": 65281,
    "usage": 4
  }
]
```
In the above output:

Both the first and second devices have the same path (DevSrvsID:4295921431) but different usagePages (12 and 11).
Similarly, devices 3, 4, and 5 share the path DevSrvsID:4295921433 but have different usagePages.
### Expected Behavior:
I would expect each device with a different usagePage to have a unique path, ensuring that node-hid can correctly distinguish between them, especially when interacting with devices like the Tilde Dongle Evo USB, which supports multiple usagePages.

### Environment:
OS: macOS 14.6.1 (23G93)
node-hid version: v3.1.2
Node.js version: v20.12.1
Additional Information:
This issue seems specific to macOS, and it causes problems when trying to interact with devices using different usagePages through node-hid.

### Steps to Reproduce:
Connect a multi-usage HID device
Use node-hid to list devices.
Observe that devices with different usagePages share the same path.
Possible Solutions:
Is it possible to assign unique paths based on both the usagePage and usage values to prevent this overlap?

todbot, 2024-10-24:
This is expected behavior.  The `path` parameter is the path to the physical device and changes depending on which USB port a device is plugged into.  

A single device can support one or more usagePage/usage pairs.   Thus, to uniquely identify a particular usage on a particular device that works no matter which port a user has it plugged into, you should search on the tuple {vendorId, productId, usagePage, usage, serialNumber} (if serial exists)

closed 2024-10-24 by todbot

JeremyXXXuuu, 2024-10-24:
> This is expected behavior.  The `path` parameter is the path to the physical device and changes depending on which USB port a device is plugged into.
>
>
>
> A single device can support one or more [...]

Thank you for the clarification regarding the path parameter and how devices are uniquely identified based on the tuple {vendorId, productId, usagePage, usage, serialNumber}.

I have a follow-up question regarding sending feature reports: If I want to send a feature report to usagePage 65280 (as in the example I provided), how will this be handled when it comes to usage 3? Is it the driver that determines whether the report is sent to usage 1 or 3 on that usagePage? If so, how does the driver distinguish between them?

todbot, 2024-10-25:
> I have a follow-up question regarding sending feature reports: If I want to send a feature report to usagePage 65280 (as in the example I provided), how will this be handled when it comes to usage [...]

Usage pages at 0xff00-0xffff (65280-65535) are called vendor-defined usages.  You will need to know what that particular device is sending or receiving.  Sometimes you can figure this out by looking at the HID Report Descriptor and learning that a particular reportId is (for example) an Input report (to the computer) of 8 bytes.  But without knowing the vendor-defined protocol, you may never get that report to be sent. 

Given that the device name you give at is "Dongle Evo USB" and a preliminary web search shows that to be generic Bluetooth, `node-hid` will not be an appropriate tool to manipulate it.  You need an OS-level driver so it's added to as a Bluetooth controller.  This may happen automatically, and you should see it as a controller under Hardware / Bluetooth / Bluetooth Controller in MacOS's System Information.   Once it's an attached controller, you would then use the protocol-specific APIs to talk to whatever device is connected to your computer through it.  So for instance, the audio APIs, if the device is headphones, printing APIs if it's a printer, etc.  Finally, if it's a BLE HID device that's not a mouse or keyboard, then `node-hid` would be usable with that device.  And `node-hid` would not be aware that there's a Bluetooth dongle all.

JeremyXXXuuu, 2024-10-31:
> Usage pages at 0xff00-0xffff (65280-65535) are called vendor-defined usages. You will need to know what that particular device is sending or receiving. Sometimes you can figure this out by looking [...]

On Windows, each usage page on my Evo dongle is given a separate path, allowing me to open them individually for their respective functions (one for dongle updates, the other for computer communication). However, on macOS, both usage pages are assigned the same path, making it impossible to open them separately.

Is there a way within node-hid or macOS's HID handling to differentiate these paths, or could node-hid be modified to expose virtual paths for each usage page? This way, I could open each one independently, even if the OS provides a single system path.

todbot, 2024-10-31:
The paths are a platform-specific way to refer to a device. Node-hid has no insight into them, as they are a reflection of the underlying `hidapi` library. 

As mentioned previously, if you want to access a specific usage on a specific device, you need to filter on the tuple {vendorId, productId, usagePage, usage, serialNumber}.  You can see one way of doing this in the [test-teensyrawhid.js](https://github.com/node-hid/node-hid/blob/master/src/test-teensyrawhid.js) example.

JeremyXXXuuu, 2024-11-01:
> The paths are a platform-specific way to refer to a device. Node-hid has no insight into them, as they are a reflection of the underlying `hidapi` library.
>
> As mentioned previously, if you want to [...]

Thank you for sharing the example! It’s helpful for understanding the filtering setup by usage page and usage.

However, I’m encountering an issue specific to macOS: even when filtering by usage page, macOS represents both usage pages with the same device path. This prevents me from opening them separately, as I can on Windows, where each usage page appears with a unique path. ( To clarify, the open method in node-hid opens a device by its path, and since macOS assigns the same path for each usage page, I'm unable to access each usage separately. This is causing issues as it prevents individual communication with different usage pages on the same device, unlike on Windows where each usage page has a unique path.)

Is there any recommended approach or workaround in node-hid or hidapi that could help distinguish and interact with the separate usage pages on macOS? Or, is there a way to implement a unique path mapping for different usage pages on macOS to facilitate separate access? Any insights would be greatly appreciated!