node-hid/node-hid #552: Unable to open device with path

AneesShaik167, 2024-11-25:
We are encountering a peculiar issue which is being faced only in macOS but not in windows. We are opening a device and storing it in map(**export const connectedDevices = new Map();**) on connecting the device. If we try to open the device again using const newDevice = new HID.HID(path); we are facing "Uncaught TypeError: cannot open device with path DevSrvsID:xxxxx" although device is available when we log const hidDevices = HID.devices();. 
  
  We also observed connectedDevices is getting empty due to some reason after some activity in the app so couldn't use the stored device hence we are opening the device again for fulfilling the operation which is working in windows.
  
  @todbot and everyone kindly help here.

Julusian, 2024-11-25:
macos only allows one open handle to a hid device at a time. windows does not have the same limitation.
So if you already have the hid device open, or another application has opened the hid device, you will not be able to open it.

AneesShaik167, 2024-11-25:
@Julusian if there is any workaround to proceed kindly let us know...

AneesShaik167, 2024-11-27:
Hi, @Julusian @todbot @everyone we are blocked here and waiting for a workaround or any solution that can help us. Kindly guide us here...