node-hid/node-hid #553: the device‘s path is  ' '

Doker-Ambition, 2024-12-05:
Hello, I would like to upload my Electron application to the MacStore, which involves a sandbox. I have noticed a strange phenomenon where when the sandbox is not enabled, I can obtain the complete device information. However, when I run in sandbox mode, all other properties can be obtained, but the path is empty, which prevents me from obtaining the object of the specific device through const newDevice=new HID HID(path)

todbot, 2024-12-05:
I'm not sure the required plist entitlements required to use `node-hid` are allowed in sandboxed apps. 

Last I looked, these were the entitlements needed: https://github.com/todbot/electron-hid-toy/blob/master/pkg/entitlements.mac.inherit.plist

Doker-Ambition, 2024-12-06:
I have checked my entities file and I have already added everything you provided before, but it still doesn't work. The path is still empty, and when I tried to use vendor ID and product ID to find the device: (new HID.HID (vid, pid)), I still couldn't find it. Even though vendor ID and product ID exist, I don't know if you have encountered this situation before. What was the path I used node-hid@3.1.2

ert78gb, 2024-12-06:
https://github.com/UltimateHackingKeyboard/agent/blob/master/scripts/entitlements.mac.plist is our plist. the difference is we have the `com.apple.security.device.usb` entry too

Doker-Ambition, 2024-12-09:
I understand why now, because my node_module package has been replaced, and it seems that there was a dependency issue that ultimately led to this phenomenon. Thank you again for your help @ert78gb、@todbot