node-hid/node-hid #554: libc++abi: terminating due to uncaught exception of type Napi::Error

icyqwq, 2024-12-21:
The Electron application on macOS crashes on exit with the following error:
libc++abi: terminating due to uncaught exception of type Napi::Error.

After some investigation, it appears that the crash only occurs when HID input is being listened to.

```
Thread 0 Crashed:: CrBrowserMain Dispatch queue: com.apple.main-thread
0   libsystem_kernel.dylib        	    0x7ff817017112 __pthread_kill + 10
1   libsystem_pthread.dylib       	    0x7ff81704d233 pthread_kill + 263
2   libsystem_c.dylib             	    0x7ff816f99d10 abort + 123
3   libc++abi.dylib               	    0x7ff81700a0b2 abort_message + 241
4   libc++abi.dylib               	    0x7ff816ffb1e5 demangling_terminate_handler() + 242
5   libobjc.A.dylib               	    0x7ff816ef8511 _objc_terminate() + 104
6   libc++abi.dylib               	    0x7ff8170094d7 std::__terminate(void (*)()) + 8
7   libc++abi.dylib               	    0x7ff81700bd55 __cxxabiv1::failed_throw(__cxxabiv1::__cxa_exception*) + 27
8   libc++abi.dylib               	    0x7ff81700bd1c __cxa_throw + 116
9   HID.node                      	       0x1e9ca32d9 Napi::Error::ThrowAsJavaScriptException() const + 281 (napi-inl.h:2409)
10  HID.node                      	       0x1e9ca7a9d napi_value__* Napi::details::WrapCallback<Napi::AsyncWorker::OnWorkComplete(Napi::Env, napi_status)::'lambda'()>(Napi::AsyncWorker::OnWorkComplete(Napi::Env, napi_status)::'lambda'()) + 21 (napi-inl.h:76) [inlined]
11  HID.node                      	       0x1e9ca7a9d Napi::AsyncWorker::OnWorkComplete(Napi::Env, napi_status) + 205 (napi-inl.h:4417)
12  Electron Framework            	       0x119e50f4a node_api_get_module_file_name + 3322
13  Electron Framework            	       0x112099c2d uv_barrier_destroy + 1245
14  Electron Framework            	       0x11209dfe7 uv_async_send + 1047
15  Electron Framework            	       0x1120b02bf uv_free_interface_addresses + 2495
16  Electron Framework            	       0x11209e565 uv_run + 485
17  Electron Framework            	       0x119e2018e node::SetCppgcReference(v8::Isolate*, v8::Local<v8::Object>, void*) + 48414
18  Electron Framework            	       0x119e20735 node::SetCppgcReference(v8::Isolate*, v8::Local<v8::Object>, void*) + 49861
19  Electron Framework            	       0x119ddf101 node::FreeEnvironment(node::Environment*) + 177
20  Electron Framework            	       0x11226edb9 v8::CodeEvent::GetScriptName() + 282985
21  Electron Framework            	       0x1121b0b5e v8::BackingStore::MaxByteLength() const + 77278
22  Electron Framework            	       0x114c1e966 v8::internal::compiler::CompilationDependencies::FieldTypeDependencyOffTheRecord(v8::internal::compiler::MapRef, v8::internal::compiler::MapRef, v8::internal::InternalIndex, v8::internal::compiler::ObjectRef) const + 11088022
23  Electron Framework            	       0x114c20170 v8::internal::compiler::CompilationDependencies::FieldTypeDependencyOffTheRecord(v8::internal::compiler::MapRef, v8::internal::compiler::MapRef, v8::internal::InternalIndex, v8::internal::compiler::ObjectRef) const + 11094176
24  Electron Framework            	       0x114c1c06e v8::internal::compiler::CompilationDependencies::FieldTypeDependencyOffTheRecord(v8::internal::compiler::MapRef, v8::internal::compiler::MapRef, v8::internal::InternalIndex, v8::internal::compiler::ObjectRef) const + 11077534
25  Electron Framework            	       0x112442472 v8::CodeEvent::GetScriptLine() + 475474
26  Electron Framework            	       0x1124435fd v8::CodeEvent::GetScriptLine() + 479965
27  Electron Framework            	       0x112443431 v8::CodeEvent::GetScriptLine() + 479505
28  Electron Framework            	       0x112441d04 v8::CodeEvent::GetScriptLine() + 473572
29  Electron Framework            	       0x112441e12 v8::CodeEvent::GetScriptLine() + 473842
30  Electron Framework            	       0x1120b0921 ElectronMain + 209
31  dyld                          	       0x10798e4fe start + 462

```

Here’s the attached complete crash log: 
[crash-report.txt](https://github.com/user-attachments/files/18219564/bugreport.txt)

Julusian, 2024-12-21:
What version of node-hid is this?
Have you attached an error handler to the opened device?

icyqwq, 2024-12-22:
Yes, I attached error handler like this:
```
deviceStatus.serialNumber = foundDevice1.serialNumber;
        hidDevice.ch0.handle = await HID.HIDAsync.open(devicePath1);
        hidDevice.ch0.handle.on('data', packetReceiverInstance.processChannel0Data.bind(packetReceiverInstance));
        hidDevice.ch0.handle.on('error', (err) => {
            console.error('USB Channel0 ERROR: ', err);
            cleanUpConnectedDevice()
        });

        hidDevice.ch1.handle = await HID.HIDAsync.open(devicePath2);
        hidDevice.ch1.handle.on('data', packetReceiverInstance.processChannel1Data.bind(packetReceiverInstance));
        hidDevice.ch1.handle.on('error', (err) => {
            console.error('USB Channel1 ERROR: ', err);
            // cleanUpConnectedDevice()
        });
```
To spped-up HID read and write, I enabled two HID channels on one device. And this crash occurs only on macOS; everything seems to work fine on Windows.

Versions:
node-hid   3.1.2
electron    31.1.0
node          v18.20.5

ert78gb, 2024-12-22:
I experience same issue with electron 32 and 33 too with MacOs 15 . I haven't experienced it MacOs 14. I thought it electron issue because I found similar issue on the electron issue tracker, but I don't find now :(

Julusian, 2024-12-24:
Could someone provide a simple repository I can clone to reproduce this with?

ert78gb, 2024-12-24:
Not simple but full public https://github.com/UltimateHackingKeyboard/agent and unfortunately need a special device for it.

@Julusian thanks for your effort.

ert78gb, 2024-12-24:
I am thinking on how could I create a simple repo

Julusian, 2024-12-24:
@ert78gb the discussion above suggests it is related to needing to to be reading from a device. so perhaps something simple which just reads from a device (assuming the device is constantly/frequently producing data) would be enough?
I have some hid devices (elgato streamdeck) which don't need any setup and produce packets whenever a key is pressed, which are my goto for any hid testing.

I would try doing it myself, but it makes it harder as I can't tell if its not happening because of my device/environment or because the test script is not hitting the right scenario.

ert78gb, 2024-12-28:
I created the https://github.com/ert78gb/node-hid-electron-exit repo. I tried to use minimal code as possible. 
You have to modify the https://github.com/ert78gb/node-hid-electron-exit/blob/29d488bb6a4e20b78b36ef3e2ae7a38dbf058976/main/device.service.js#L138 to communicate with your device.

The master branch is not crashing. Few times when exiting the electron I could catch the `Cannot write to hid device` exception. 

While I created the reproduction repo I recognised I use a few extra `Promises` that makes the event loop more busy. The `chore-promises` branch contains this version. This branch crashes very rarely. Maybe I have to add more Promises to simulate our real application and crashes. I found only a closed electron issue that describe the same issue https://github.com/electron/electron/issues/26880

ert78gb, 2025-08-31:
I dont have. We still facing with this issue with electron 37 too