node-hid/node-hid #557: Upgrade napi version to make node-hid compatible with node lts/jod

hui-an-yang, 2025-03-20:
Hello

The issue we encountered originating from node lts/Jod(22.14.0) and above when run `npm ci` in [pr ci script](https://github.com/ecadlabs/taquito/actions/runs/13976674911/job/39132086929) will get error `libusb.h: No such file or directory` in linux machine. 
So I created [issue#57470](https://github.com/nodejs/node/issues/57470) in nodejs repo and their reply is: "node v22.14.0 introduces napi v10. That node-hid module you're trying to install doesn't have precompiled packages for that version, tries to compile from source, and fails because of missing header files."
In my repository I am using `@ledgerhq/hw-transport-node-hid-noevents` which use node-hid v2.1.2 I have a pr to manually upgreade node-hid to v3.1.2 but the ci failure persist.
I am wondering what's the timeline for node-hid to release a new version with napi_version 10?
I understand node-hid officially support to node v20(maintenance) for now, but v22 is the current LTS
I am hoping raising this now will let you consider migrating to newer napi earlier.
Thank you!

[image]

Julusian, 2025-03-20:
Oh ok, nodejs have changed the forward compatibility guarantees of the node-api, so I need to re-read that and figure out what that means.

But the current builds do work with node v22.14.

> node v22.14.0 introduces napi v10. That node-hid module you're trying to install doesn't have precompiled packages for that version, tries to compile from source, and fails because of missing header files.

I think this is happening because of a bug in the library which is installing the native binary. The fix was https://github.com/prebuild/prebuild-install/pull/204, so try forcing `prebuild-install` to update to v7.1.3 or later, hopefully that will resolve it.

I don't expect a new release of v2 of this library to happen, but if this is in fact a problem in v3.1.2 then I will look into a fix

Julusian, 2025-03-20:
From https://nodejs.org/api/n-api.html#node-api-version-matrix
>  Existing add-ons can continue to run without recompilation using an earlier version of Node-API.

So we shouldn't need to do anything, as we are compiling for v4, and that page says that it "can continue to run without recompilation". 

I am pretty sure this is simply a bug in the library we used to use which installs the prebuilds, and it not doing the install when expected.

hui-an-yang, 2025-03-20:
Hello @Julusian
 
Thanks for getting back to me and point me direction to have this resolved 🙌
What eventually fix it for me is forcing `napi-build-tuils` to be v2.0.0, feel free to close this issue anytime :)

Julusian, 2025-06-03:
Reading this back, I don't think there is anything we need to do on this. Let me know if you disagree