node-hid/node-hid #562: Build error

chuan-1024, 2025-08-04:
`[2025-08-04T06:19:33.192Z] ../src/HID.cc:612:16: error: cannot convert ‘void (*)(void*)’ to ‘node::Environment*’
[2025-08-04T06:19:33.192Z]    node::AtExit(deinitialize, 0);
[2025-08-04T06:19:33.192Z]                 ^~~~~~~~~~~~
[2025-08-04T06:19:33.192Z] In file included from ../src/HID.cc:30:
[2025-08-04T06:19:33.192Z] /home/jenkins/.cache/node-gyp/16.20.2/include/node/node.h:979:38: note:   initializing argument 1 of ‘void node::AtExit(node::Environment*, void (*)(void*), void*)’
[2025-08-04T06:19:33.192Z]  NODE_EXTERN void AtExit(Environment* env,`

Julusian, 2025-08-04:
What version of node-hid are you using?

Julusian, 2025-08-04:
> 3.1.2

I don't think it is, based on that build error, I am going to guess you are on v0.7 of node-hid, the line numbers look to line up with that.
In 3.1.2, `../src/HID.cc:612:16` isnt a valid line number

The latest 0.7 release is over 6 years old now, so I am not surprised it has build issues. The latest release should be a drop in replacement, and brings a bunch of benefits including not needing to compile it.

So I suggest making sure it is installing the correct version, and that you dont have multiple versions being installed (which is easily done, there are a bunch of packages on npm which are a bit stale and are using an equally old node-hid)

chuan-1024, 2025-08-04:
> > 3.1.2
> 
> 我不这么认为。根据那个构建错误，我猜你用的 node-hid v0.7 版本，行号看起来和 v0.7 一致。 在 3.1.2 版本中，`../src/HID.cc:612:16`行号无效。
> 
> 最新的 0.7 版本已经发布六年多了，所以它出现构建问题我一点也不意外。最新版本应该可以作为替代品，而且有很多好处，包括无需编译。
> 
> 所以我建议确保它安装了正确的版本，并且没有安装多个版本（这很容易做到，npm 上有很多包有点陈旧并且使用同样旧的 node-hid）

Thanks for the answer. I will try version 3.2.0.

closed 2025-09-01 by Julusian